rggen/rggen #17: type of registers supported

dev0075, 2020-01-05:
Hi taichi-ishitani,
      I have come across your rggen for generating registers, Can you please tell me the type of registers it supports. Is it mentioned in any of your docs ? Please let me know. If it's not mentioned tell me here and I will add that in your docs. 

Thank You,

taichi-ishitani, 2020-01-06:
Hi @dev0075 ,
Thank you for opening this issue !
You can get type of registers and bit fields supported by RgGen from following wiki documents.

* [type of registers](https://github.com/rggen/rggen/wiki/Register-Map-Specifications#type)
* [type of bit fields](https://github.com/rggen/rggen/wiki/Register-Map-Specifications#type-1)

dev0075, 2020-01-06:
Hi @taichi-ishitani ,

Thank you for the swift reply ! 
This is great help !
Thanks !

closed 2020-01-06 by dev0075

taichi-ishitani, 2020-01-06:
Hi @dev0075 ,

I'd like to increase users of RgGen so I have some questions.

1. How did you find RgGen ?
2. Can you recommend me where I should post an article to introduce RgGen or communities for RTL engineers?

dev0075, 2020-01-06:
Hi @taichi-ishitani 
1. I was searching for register gen tools on Google and stumbled on your tools. 
2. You can contact popular website teams like eetimes, semiengineering.com, design-reuse.com, if you haven't put this ok opencores.org, please do it's popular among engineers. Your tool is awesome and  I'm pretty sure RTL engineers like myself would love to use and recommend. 

Thanks !