rggen/rggen #31: Could you give me some ideas about rggen scripts？

zhajio1988, 2020-03-17:
Hi @taichi-ishitani 
It‘s amazing  to see your scripts, but i am not familiar with RUBY, if i rewrite this scripts to python,
could you give me some ideas about how do you write rggen and your thinking?

taichi-ishitani, 2020-03-17:
Hi @zhajio1988 ,
Thank you for your interest in RgGen !
What kind of information do you want?

taichi-ishitani, 2020-03-17:
What really do you want to do?
Do you want to extend RgGen to support your special bit field types or host interface protocols?
If yes, it's very easy and you can do that even though you are not familiar with ruby, I think.

zhajio1988, 2020-03-30:
@taichi-ishitani i want to implement something like auto connection when SOC integration. If your scripts can do this, could you give me some ideas?

taichi-ishitani, 2020-03-30:
@zhajio1988 ,
What does 'auto connection' mean?
Do you think something like IP-XACT?

zhajio1988, 2020-03-31:
@yes or no，may be something like yaml config file？

taichi-ishitani, 2020-04-02:
@zhajio1988 ,
I don't understand what you want to do.
To get good understandings, I think we should communicate more interactively.
How about using Gitter chat?
https://gitter.im/rggen/rggen