rggen/rggen #32: DUH support

drom, 2020-03-18:
Great project.
We are developing a new format named DUH. It can capture register information.
Would you be interested in adding D∪H format support?

## more info

* https://github.com/sifive/duh
* https://github.com/sifive/duh-schema/blob/master/lib/register.js
* https://github.com/sifive/block-pio-sifive/blob/master/docs/pio.json5#L94

taichi-ishitani, 2020-03-19:
Hi @drom ,

Thank you for your opening this ticket !
I'm interested in supporting DUH format to increase number of users.

I looked int to the sample JSON file and I think its format is very simple.
Therefore, I think it is not difficult to create a loader component for DUH format.
I have some questions about DUH format.

* What access type does DUT support?
    * read-write/read-only/write-only?
    * How about other access types such as W1C (write 1 to clear/read only) ?
* RgGen does not support parameterized register width and bit field width.
    * Can you accept the limitation that these values should be constant value?
    * Do other values which can be parameterized exist?

drom, 2020-03-19:
https://json5.org
https://github.com/bartoszkopinski/json5
access - https://github.com/sifive/duh-schema/blob/master/lib/primitive.js#L35
modifiedWriteValue - https://github.com/sifive/duh-schema/blob/master/lib/field.js#L6
fixed field width is fine.

taichi-ishitani, 2020-03-19:
@drom ,
Thank you for your update.
Can you give me the priority of this ticket?
I'm planning to update RgGen to introduce 'register file' layer.
If you're not in hurry I'd like to prioritize introducing 'register file' layer.

taichi-ishitani, 2020-07-23:
Hi @drom ,

I have a question.

https://github.com/sifive/block-pio-sifive/blob/36fbe0d07c599ce700b5471e511f64ddb06c15b0/docs/pio.json5#L103
In the above sample, width of register `ODATA` is set to `pioWidth`. How can I get actual value of `pioWidth` ? Is this a Verilog parameter?

drom, 2020-07-23:
yes, in the component, it is a constraint (in form of JSON schema). It will be passed down to the Verilog parameter.
The actual value will be decided at instantiation time.

taichi-ishitani, 2020-07-23:
Thanks. I understood.

One more question.
https://github.com/sifive/duh-schema/blob/86e26211117b514b18d006cbd0d6c4fe1931bee3/lib/register.js#L16
Is type of `pioWidth` `expression` of the register schema?

drom, 2020-07-23:
Yes, some of the register/field properties can depend on parameters.
Simple math expressions like `a * 8` can be used to relate them to parameters.

taichi-ishitani, 2020-07-24:
Thanks, I understood.
However, RgGen does not support parameterized bit width, register width, etc. because RgGen needs fixed these values to validate input register maps.

taichi-ishitani, 2020-08-01:
Hi @drom ,

I have questions.

1.
https://github.com/sifive/duh/blob/master/docs/component.md#registers
In this example `memoryMaps` is a member of `component`.
https://github.com/sifive/block-pio-sifive/blob/master/docs/pio.json5
But in this example, it is a member of `definition`.
What is different between these `memoryMaps`.

2.
Do you have an example DUT file including `registerFIle` items?