rggen/rggen #45: How to compile generated systemverilog?

zzeng00, 2020-08-24:
Hello, I have a question about synthesis/compiling.
I tried to compile the sample systemverilog file block_0.sv by using VCS. And it suggested me I also need rggen_rtl_pkg. Then I downloaded the whole rggen-sv-rtl-master, and tried to compile block_0.sv with compile.f. 

But then the problems come out:
**Top Level Modules:
       rggen_bit_field_rc
       rggen_bit_field_reserved
       rggen_bit_field_ro
       rggen_bit_field_rs
       rggen_bit_field_rw_wo
       rggen_bit_field_rwc
       rggen_bit_field_rwe
       rggen_bit_field_rwl
       rggen_bit_field_rws
       rggen_bit_field_w01c_wc_woc
       rggen_bit_field_w01crs_wcrs
       rggen_bit_field_w01s_ws_wos
       rggen_bit_field_w01t
       rggen_bit_field_w01src_wsrc
       rggen_bit_field_w01trg
       rggen_bit_field_wrc
       rggen_bit_field_wrs
       rggen_default_register
       rggen_external_register
       rggen_indirect_register
       rggen_apb_adapter
       rggen_apb_bridge
       rggen_axi4lite_adapter
       rggen_axi4lite_bridge

Error-[SV-UIP] Unconnected interface port
./rggen-sv-rtl-master/rggen_bit_field_rc.sv, 7
"bit_field_if"
  The port 'bit_field_if' of top-level module 'rggen_bit_field_rc' whose type 
  is interface 'rggen_bit_field_if' is left unconnected. It is illegal to 
  leave the interface ports unconnected.
  Please make sure that all the interface ports are connected.

Error-[SV-UIP] Unconnected interface port
./rggen-sv-rtl-master/rggen_bit_field_reserved.sv, 2
"bit_field_if"
  The port 'bit_field_if' of top-level module 'rggen_bit_field_reserved' whose
  type is interface 'rggen_bit_field_if' is left unconnected. It is illegal to
  leave the interface ports unconnected.
  Please make sure that all the interface ports are connected.

Error-[SV-UIP] Unconnected interface port
./rggen-sv-rtl-master/rggen_bit_field_ro.sv, 4
"bit_field_if"
  The port 'bit_field_if' of top-level module 'rggen_bit_field_ro' whose type 
  is interface 'rggen_bit_field_if' is left unconnected. It is illegal to 
  leave the interface ports unconnected.
  Please make sure that all the interface ports are connected.

Error-[SV-UIP] Unconnected interface port
./rggen-sv-rtl-master/rggen_bit_field_rs.sv, 7
"bit_field_if"
  The port 'bit_field_if' of top-level module 'rggen_bit_field_rs' whose type 
  is interface 'rggen_bit_field_if' is left unconnected. It is illegal to 
  leave the interface ports unconnected.
  Please make sure that all the interface ports are connected.

Error-[SV-UIP] Unconnected interface port
./rggen-sv-rtl-master/rggen_bit_field_rw_wo.sv, 9
"bit_field_if"
  The port 'bit_field_if' of top-level module 'rggen_bit_field_rw_wo' whose 
  type is interface 'rggen_bit_field_if' is left unconnected. It is illegal to
  leave the interface ports unconnected.
  Please make sure that all the interface ports are connected.

Error-[SV-UIP] Unconnected interface port
./rggen-sv-rtl-master/rggen_bit_field_rwc.sv, 8
"bit_field_if"
  The port 'bit_field_if' of top-level module 'rggen_bit_field_rwc' whose type
  is interface 'rggen_bit_field_if' is left unconnected. It is illegal to 
  leave the interface ports unconnected.
  Please make sure that all the interface ports are connected.

Error-[SV-UIP] Unconnected interface port
./rggen-sv-rtl-master/rggen_bit_field_rwe.sv, 7
"bit_field_if"
  The port 'bit_field_if' of top-level module 'rggen_bit_field_rwe' whose type
  is interface 'rggen_bit_field_if' is left unconnected. It is illegal to 
  leave the interface ports unconnected.
  Please make sure that all the interface ports are connected.**

I found that you are using interface and modport to define these "unconnected ports". 
Since I am not very familiar with this syntax, I am not sure if it is synthesizable. 
And I am also wondering whether my script is correct or not.
Below is my run_vcs script:
**source /home/zzeng00/.bashrc
vcs -sverilog -f file.f -R -debug_all -timescale=1ns/10ps +lint=TFIPC-L +lint=PCWM -y /tools/eda/dc2016-L-2016.03-SP1/dw/sim_ver +incdir+/tools/eda/dc2016-L-2016.03-SP1/dw/sim_ver +libext+.v**

Could you point out my mistake? 
Thank you very much!

taichi-ishitani, 2020-08-24:
Hi @zzeng00 ,

I think there are 2 ways to fix this problem:

1. Specify your top module by using `-top` compile option
    - Example: https://github.com/rggen/rggen-sample-testbench/blob/818e26e8db9ef022021ef0d3d53ddf228bdb9ec6/sim/vcs.mk#L14
2. Remove reported modules from your file list
    - I think these modules are not used withing generated CSR module so you don't need to compile these modules

taichi-ishitani, 2020-08-24:
Above answer may be wrong.
Can you put the log file? I need it to investigate the error.

taichi-ishitani, 2020-08-24:
Did you add generated RTL to your file list?

zzeng00, 2020-08-24:
Hi tachi!
thanks for your reply!
I am out of office now.  I added the generated rtl to filelist at first,
and then removed it, Same issue happens.

I will post my log file tomorrow.
Let's fix this problem!
Thx!

taichi-ishitani, 2020-08-24:
```
vcs \
-full64 \
-sverilog \
-l vcs.log \
+incdir+/home/ishitani/workspace/rggen-sample-testbench/env \
+define+RGGEN_ENABLE_BACKDOOR \
+define+RGGEN_ENABLE_SVA \
+define+RGGEN_ENABLE_ENHANCED_RAL \
-timescale=1ns/1ps -ntb_opts uvm-1.2 \
-top top \
-f /home/ishitani/workspace/rggen-sample-testbench/env/tue/compile.f \
-f /home/ishitani/workspace/rggen-sample-testbench/rtl/rggen-sv-rtl/compile.f \
-f /home/ishitani/workspace/rggen-sample-testbench/ral/rggen-sv-ral/compile.f \
-f /home/ishitani/workspace/rggen-sample-testbench/env/tvip-apb/compile.f \
/home/ishitani/workspace/rggen-sample-testbench/ral/block_0_ral_pkg.sv \
/home/ishitani/workspace/rggen-sample-testbench/ral/block_1_ral_pkg.sv \
/home/ishitani/workspace/rggen-sample-testbench/env/env_pkg.sv \
/home/ishitani/workspace/rggen-sample-testbench/rtl/apb/block_0.sv \
/home/ishitani/workspace/rggen-sample-testbench/rtl/apb/block_1.sv \
/home/ishitani/workspace/rggen-sample-testbench/env/apb_env_pkg.sv \
/home/ishitani/workspace/rggen-sample-testbench/env/apb_top.sv
```

This command is to compile the sample UVM test bench.
https://github.com/rggen/rggen-sample-testbench
I think this can help you to understand how to generated RTL and supplemental modules.

zzeng00, 2020-08-25:
Hello, I still could not solve the problem.
I am still trying to compile block_0.sv,  below is the filelist:
./rggen_rtl_pkg.sv
./rggen_apb_if.sv
./rggen_apb_adapter.sv
./rggen_register_if.sv
./rggen_bus_if.sv
./block_0.sv
I put them in the same folder.
And then I tried to compile it. The same issues happened:

Parsing design file './rggen_rtl_pkg.sv'
Parsing design file './rggen_apb_if.sv'
Parsing design file './rggen_apb_adapter.sv'
Parsing design file './rggen_register_if.sv'
Parsing design file './rggen_bus_if.sv'
Parsing design file './block_0.sv'
Top Level Modules:
       block_0

Error-[SV-UIP] Unconnected interface port
./block_0.sv, 22
"apb_if"
  The port 'apb_if' of top-level module 'block_0' whose type is interface 
  'rggen_apb_if' is left unconnected. It is illegal to leave the interface 
  ports unconnected.
  Please make sure that all the interface ports are connected.

Error-[SV-UIP] Unconnected interface port
./block_0.sv, 82
"register_12_bus_if"
  The port 'register_12_bus_if' of top-level module 'block_0' whose type is 
  interface 'rggen_bus_if' is left unconnected. It is illegal to leave the 
  interface ports unconnected.
  Please make sure that all the interface ports are connected.

2 errors
CPU time: .175 seconds to compile

zzeng00, 2020-08-25:
I want to follow the instruction on https://github.com/rggen/rggen-sample-testbench, but I do not have permission to set the export path on the machine in my office.......

taichi-ishitani, 2020-08-25:
The top module cannot have SV interface port so these errors happened.
Can you try to compile again by using the top module below and command?

Top module:

```systemverilog
// top.sv
module top;
  timeunit  1ns/1ps;

  logic                       clk;
  logic                       rst_n;
  rggen_apb_if #(16, 32)      apb_if();
  logic [3:0]                 register_0_bit_field_0;
  logic [3:0]                 register_0_bit_field_1;
  logic                       register_0_bit_field_2;
  logic [1:0]                 register_0_bit_field_3;
  logic [1:0]                 register_0_bit_field_4;
  logic [1:0]                 register_0_bit_field_5;
  logic                       register_1;
  logic [3:0]                 register_2_bit_field_0;
  logic [3:0]                 register_2_bit_field_1;
  logic [3:0]                 register_3_bit_field_0;
  logic [3:0]                 register_3_bit_field_1;
  logic [3:0]                 register_3_bit_field_2_trigger;
  logic [3:0]                 register_3_bit_field_3_trigger;
  logic [3:0]                 register_4_bit_field_0_set;
  logic [3:0]                 register_4_bit_field_0;
  logic [3:0]                 register_4_bit_field_1_set;
  logic [3:0]                 register_4_bit_field_1;
  logic [3:0]                 register_4_bit_field_1_unmasked;
  logic [3:0]                 register_4_bit_field_3_clear;
  logic [3:0]                 register_4_bit_field_3;
  logic                       register_5_bit_field_0_clear;
  logic [1:0]                 register_5_bit_field_0;
  logic [1:0]                 register_5_bit_field_1;
  logic                       register_5_bit_field_2_set;
  logic [1:0]                 register_5_bit_field_2[2];
  logic [1:0]                 register_5_bit_field_3[2];
  logic                       register_5_bit_field_4_enable;
  logic [1:0]                 register_5_bit_field_4;
  logic [1:0]                 register_5_bit_field_5;
  logic [1:0]                 register_5_bit_field_6;
  logic                       register_5_bit_field_7_lock;
  logic [1:0]                 register_5_bit_field_7;
  logic [1:0]                 register_5_bit_field_8;
  logic [1:0]                 register_5_bit_field_9;
  logic [3:0]                 register_6_bit_field_0_set;
  logic [3:0]                 register_6_bit_field_0;
  logic [3:0]                 register_6_bit_field_1_set;
  logic [3:0]                 register_6_bit_field_1;
  logic [3:0]                 register_6_bit_field_1_unmasked;
  logic [3:0]                 register_6_bit_field_3_set;
  logic [3:0]                 register_6_bit_field_3;
  logic [3:0]                 register_6_bit_field_4_set;
  logic [3:0]                 register_6_bit_field_4;
  logic [3:0]                 register_6_bit_field_4_unmasked;
  logic [3:0]                 register_6_bit_field_6_clear;
  logic [3:0]                 register_6_bit_field_6;
  logic [3:0]                 register_6_bit_field_7_clear;
  logic [3:0]                 register_6_bit_field_7;
  logic [3:0]                 register_6_bit_field_8;
  logic [3:0]                 register_6_bit_field_9;
  logic [3:0]                 register_7_bit_field_0;
  logic [3:0]                 register_7_bit_field_1;
  logic [3:0]                 register_7_bit_field_2;
  logic [3:0]                 register_7_bit_field_3;
  logic [3:0]                 register_8_bit_field_0_set;
  logic [3:0]                 register_8_bit_field_0;
  logic [3:0]                 register_8_bit_field_1_clear;
[... 122 lines cut ...]
    .o_register_8_bit_field_1           (register_8_bit_field_1           ),
    .i_register_8_bit_field_2_set       (register_8_bit_field_2_set       ),
    .o_register_8_bit_field_2           (register_8_bit_field_2           ),
    .i_register_8_bit_field_3_clear     (register_8_bit_field_3_clear     ),
    .o_register_8_bit_field_3           (register_8_bit_field_3           ),
    .o_register_8_bit_field_4           (register_8_bit_field_4           ),
    .o_register_8_bit_field_5           (register_8_bit_field_5           ),
    .o_register_9_bit_field_0           (register_9_bit_field_0           ),
    .o_register_9_bit_field_1           (register_9_bit_field_1           ),
    .o_register_9_bit_field_2           (register_9_bit_field_2           ),
    .o_register_10_bit_field_0          (register_10_bit_field_0          ),
    .o_register_10_bit_field_1          (register_10_bit_field_1          ),
    .o_register_11_bit_field_0          (register_11_bit_field_0          ),
    .o_register_12_bit_field_0          (register_12_bit_field_0          ),
    .register_13_bus_if                 (register_13_bus_if               )
  );
endmodule
```

Command

```
$ export RGGEN_SV_RTL_ROOT=PATH_TO_RGGEN_SV_RTL
$ vcs -full64 -sverilog -top top -f ${RGGEN_SV_RTL_ROOT}/compile.f block_0.sv top.sv
```

zzeng00, 2020-08-25:
Thanks!
top module cannot have SV interface port ! This is the problem!
I have solved it.

taichi-ishitani, 2020-08-25:
Good!
Can you close this issue if you have no topics?

closed 2020-08-26 by zzeng00