rggen/rggen #73: Change notation of bit assignment

taichi-ishitani, 2021-02-25:
Change notation of bit assignment to swap position of LSB and width.
This is to support bit assignment without LSB position on XLSX register map.

For backward compatibility, add a new configuration parameter to specify what is the first element of notation of bit assignment.

closed 2021-02-25 by taichi-ishitani

cookacounty, 2021-02-28:
The SystemRDL spec is probably the best standard to follow:

https://www.accellera.org/images/downloads/standards/systemrdl/SystemRDL_2.0_Jan2018.pdf

[image]

You do not need to explicitly set the LSB (its assumed you start at zero).

For example if I defined some fields

field_a [4]
field_b [4]

I would get registers

field_a at bits [3:0] and field_b at bits [7:4]

This makes it pretty nice because you can just put in the width of the fields you need and then the tool keeps track of the indexes. Then when generated the tool puts out the physical indexes and addresses.

taichi-ishitani, 2021-02-28:
@cookacounty ,
Thank you for your comment!
Now, you can omit LSB specifications when you use register map specification written in Excel.

[image]

For this case, bit assignments of these bit fields are as followings:

* bit_field_0: [3:0]
* bit_field_1: [7:4]
* bit_field_2: [8]