rggen/rggen #78: Documentation on register map semantics.

m-kru, 2021-05-23:
Is there any documentation on the register map semantics?  The README points to the repository with examples https://github.com/rggen/rggen-sample, but not all things are clear solely on the examples. For instance, what is the exact difference between the `register_block` and `register_file`? Is it possible to create nested structure of the `register_block`s?

taichi-ishitani, 2021-05-23:
Hi @m-kru ,
Thank you for opening this ticket!

> Is there any documentation on the register map semantics?

Please see the wiki document below.
https://github.com/rggen/rggen/wiki/Register-Map-Specifications

m-kru, 2021-05-24:
@taichi-ishitani Ok, so it looks like `register_block`s can not be nested, however `register_file`s can. What happens with such nested `register_file`s? Are they flattened or not? If not, does rggen generate all the needed bus hierarchy automatically? I mean bus infrastructure, crossbars etc. I do not care about the target language, I can choose the one with the best rggen support.

taichi-ishitani, 2021-05-24:
> Are they flattened or not?

Yes, register file layers are flattened. 
I think this sample can help you to understand.
https://github.com/rggen/rggen-sample/blob/0f28ca3cdb7781dc87bf4b09546aa0ba4b7e16a4/block_1.sv#L244

m-kru, 2021-05-24:
This bock_1.sv file is auto generated, isn't it?

taichi-ishitani, 2021-05-24:
This is auto generated file.

m-kru, 2021-05-24:
Ok, I see what is going on. Do not you think such approach is a bit inconvenient for big designs? Flattening and having single module with registers might be convenient for small design, but after crossing some complexity point, you have to propagate a lot of signals with long names via multiple modules. What is more, if you have multiple masters, single master blocks whole register tree. Does rggen support multiple masters? How do you cope with all these issues?

taichi-ishitani, 2021-05-24:
Do you want RgGen to support bus fabric generation?

m-kru, 2021-05-24:
I would not say that I currently "want" something. I am just analyzing and I am wondering how you cope with issues that I have (and probably all of us have) while dealing with buses and registers. My experience is that some register generation solutions support also bus fabric generation, while others flatten everything. The latter are easier to reason about, but are rather limited  to simple designs. I am trying to establish to which group the rggen belongs, or maybe it lies somewhere in between.

taichi-ishitani, 2021-05-24:
ReGen generate CSR modules per register block.
If you input two register blocks (block_0 and block_1) then RgGen will generate two CSR module (block_0.sv and block_1.sv).
Register files and registers within a register block are flattened they are no combined with register files and registers which belong to other register blocks.

m-kru, 2021-05-24:
But when I generate multiple blocks, I still have to generate proper bus fabric on my own, right?

taichi-ishitani, 2021-05-24:
Yes, you need to implement bus fabric.
I think it is difficult that RgGen supports bus fabric generation because structure of bus fabric is depend on each designs.

m-kru, 2021-05-25:
The complexity depends on the bus. For Wishbone it is actually relatively easy. For AXI it might be a bit harder. Nevertheless, it is doable. The register description format just needs to support this. In case of RgGen it would have to allow nesting blocks.

taichi-ishitani, 2021-05-25:
Do you want to a wrapper module containing multiple CSR modules like below?
If yes, you can make RgGen to generate such module by developing your own plugin.
Internally, RgGen hsa a layer to collect all given register blocks.

```
+--------------------------------+
| CSR wrapper                    |
|+-------------+     +--------+  |
|| CSR block 0 |<--->|        |<-|-> Bus master
|+-------------+     |        |  |
|                    |        |  |
|+-------------+     |        |  |
|| CSR block 1 |<--->| Fabric |  |
|+-------------+     |        |  |
|                    |        |  |
|+-------------+     |        |  |
|| CSR block 2 |<--->|        |<-|-> Bus master
|+-------------+     +--------+  |
+--------------------------------+
```

m-kru, 2021-05-25:
Something list this, but with blocks nested in blocks. In such case one needs crossbars also in blocks.

taichi-ishitani, 2021-05-25:
I'm confused.
What is difference between 

* one CSR module containing all registers
* one wrapper module containing multiple CSR blocks

I think there are no differences between these modules superficially if a CSR module can have multiple bus port.
Can you explain more details?

taichi-ishitani, 2021-05-26:
You think that introducing nested register blocks is to increase maintainability of register map specifications, right?

m-kru, 2021-05-26:
You example is to simple. I am taking about multi level nesting. In such case the design has better maintainability. There is no need to propagate multiple signals via multiple modules.

taichi-ishitani, 2021-05-26:
How about this?
We need to add "register block" name to port name as prefix. This is to make port names to be unique.

```yaml
register_blocks:
  - name: innter_block_0_0
    byte_size: 128
    # regsiter descriptions

  - name: innter_block_0
    byte_size: 128
    register_blocks:
      - { name: innter_block_0_0, offset_address: 0x00 }

  - name: innter_block_1
    byte_size: 128
    # regsiter descriptions

  - name: top_block
    byte_size: 256
    register_blocks:
      - { name: innter_block_0, offset_address: 0x00 }
      - { name: innter_block_1, offset_address: 0x80 }
```

```systemverilog
module inner_block_0_0 (
  output  o_register_0_bit_field_0
);
endmodule

module inner_block_0 (
  output  o_innter_block_0_0_register_0_bit_field_0
);
  inner_block_0_0 u_innter_block_0_0 (
    .o_register_0_bit_field_0 (o_innter_block_0_0_register_0_bit_field_0  )
  );
endmodule

module inner_block_1 (
  output  o_register_0_bit_field_0
);
endmodule

module top_block (
  output  o_innter_block_0_innter_block_0_0_register_0_bit_field_0,
  output  o_innter_block_1_register_0_bit_field_0
);
  inner_block_0 u_innter_block_0 (
    .o_innter_block_0_0_register_0_bit_field_0  (o_innter_block_0_innter_block_0_0_register_0_bit_field_0 )
  );
  inner_block_1 u_innter_block_1  (
    .o_register_0_bit_field_0 (o_innter_block_1_register_0_bit_field_0  )
  );
endmodule
```

m-kru, 2021-05-27:
I feel like you are going into wrong direction. Long names of ports is exactly one of the things I want to avoid. Do not bother to implement it. I was just curious what RgGen has to offer.

taichi-ishitani, 2021-05-27:
I understand your thought.  However, it is too bother to implement

* checking uniqueness of port name
* uniquifying port name when multiple registers have the same port name

Therefore, I prefer adding block name prefix to port names to simplify implementation.

taichi-ishitani, 2021-05-28:
I've create a ticket for nested register block support #80 .
I'm also working for SystemRDL support so nested register block support has lower priority than SystemRDL support.

Please close this ticket if you have no further topics .

closed 2021-05-28 by m-kru