rggen/rggen #83: Increase maximum block size to greater than 64KB

ifthikars, 2021-07-26:
Hi,

I am using your tool to create a RAL model and I get an error that the input byte size 65540 is greater than the maximum byte size  65536.  Is there a way to increase the maximum byte size to larger than 64KB?  Please see the log below.

Best Regards,
Ifthikar

[image]

taichi-ishitani, 2021-07-26:
Hi @ifthikars ,
Thanks for using RgGen!

> Is there a way to increase the maximum byte size to larger than 64KB?

You need to edit your config file and increase the `address_width` value.
https://github.com/rggen/rggen-sample/blob/0f28ca3cdb7781dc87bf4b09546aa0ba4b7e16a4/config.yml#L2

ifthikars, 2021-07-27:
That worked.  Thanks for the prompt reply.

closed 2021-07-27 by ifthikars