rggen/rggen #105: Plugin for C header file generator

taichi-ishitani, 2022-06-07:
Create a new plugin for C header file generator.
Generated header file includes:

* macro definitions
    * for register
        * offset address
        * byte size
    * for bit field
        * bit position
        * width
        * mask
* typedefs
    * struct containing registers
        * https://github.com/taichi-ishitani/rggen/blob/411fff9f6618ffdbdc59bcb45a2d6bf39a7e40d5/sample/sample_0.h#L4

There are no limitations for bit field/register width; a bit field of which width is > 64 bits is allowed.
It is difficult to support such case completely so an error will be raised for this case.

taichi-ishitani, 2022-06-08:
Pair of write-only and read-only registers and indirect registers can be mapped into the same address range.
Mapping these registers to C struct is needed more concern and I have no idea how to do that.
Therefore, for the first version, the plugin generates macro definitions only.

taichi-ishitani, 2022-06-08:
This is a sample C header file which this plugin would generate.

```c
#ifndef BLOCK_0_H
#define BLOCK_0_H

#define REGISTER_0_BIT_FIELD_0_WIDTH 4
#define REGISTER_0_BIT_FIELD_0_MASK 0xf
#define REGISTER_0_BIT_FIELD_0_OFFSET 0

#define REGISTER_0_BIT_FIELD_1_WIDTH 4
#define REGISTER_0_BIT_FIELD_1_MASK 0xf
#define REGISTER_0_BIT_FIELD_1_OFFSET 4

#define REGISTER_0_BIT_FIELD_2_WIDTH 1
#define REGISTER_0_BIT_FIELD_2_MASK 0x1
#define REGISTER_0_BIT_FIELD_2_OFFSET 8

#define REGISTER_0_BIT_FIELD_3_WIDTH 2
#define REGISTER_0_BIT_FIELD_3_MASK 0x3
#define REGISTER_0_BIT_FIELD_3_OFFSET 9

#define REGISTER_0_BIT_FIELD_4_WIDTH 2
#define REGISTER_0_BIT_FIELD_4_MASK 0x3
#define REGISTER_0_BIT_FIELD_4_OFFSET 11

#define REGISTER_0_BIT_FIELD_5_WIDTH 2
#define REGISTER_0_BIT_FIELD_5_MASK 0x3
#define REGISTER_0_BIT_FIELD_5_OFFSET 13

#define REGISTER_0_BIT_FIELD_6_WIDTH 2
#define REGISTER_0_BIT_FIELD_6_MASK 0x3
#define REGISTER_0_BIT_FIELD_6_OFFSET 15

#define REGISTER_0_BYTE_WIDTH 4
#define REGISTER_0_BYTE_SIZE 4
#define REGISTER_0_OFFSET 0x0

#define REGISTER_1_REGISTER_1_WIDTH 1
#define REGISTER_1_REGISTER_1_MASK 0x1
#define REGISTER_1_REGISTER_1_OFFSET 0

#define REGISTER_1_BYTE_WIDTH 4
#define REGISTER_1_BYTE_SIZE 4
#define REGISTER_1_OFFSET 0x4

#define REGISTER_2_BIT_FIELD_0_WIDTH 4
#define REGISTER_2_BIT_FIELD_0_MASK 0xf
#define REGISTER_2_BIT_FIELD_0_OFFSET 0

#define REGISTER_2_BIT_FIELD_1_WIDTH 4
#define REGISTER_2_BIT_FIELD_1_MASK 0xf
#define REGISTER_2_BIT_FIELD_1_OFFSET 8

#define REGISTER_2_BIT_FIELD_2_WIDTH 8
#define REGISTER_2_BIT_FIELD_2_MASK 0xff
#define REGISTER_2_BIT_FIELD_2_OFFSET 16

#define REGISTER_2_BIT_FIELD_2_WIDTH 8
#define REGISTER_2_BIT_FIELD_2_MASK 0xff
#define REGISTER_2_BIT_FIELD_2_OFFSET 24

#define REGISTER_2_BYTE_WIDTH 4
#define REGISTER_2_BYTE_SIZE 4
#define REGISTER_2_OFFSET 0x8

#define REGISTER_3_BIT_FIELD_0_WIDTH 4
#define REGISTER_3_BIT_FIELD_0_MASK 0xf
#define REGISTER_3_BIT_FIELD_0_OFFSET 0

[... 243 lines cut ...]
#define REGISTER_11_OFFSET_2 0x60
#define REGISTER_11_OFFSET_3 0x68

#define REGISTER_12_BIT_FIELD_0_WIDTH 1
#define REGISTER_12_BIT_FIELD_0_MASK 0x1
#define REGISTER_12_BIT_FIELD_0_OFFSET 0

#define REGISTER_12_BYTE_WIDTH 4
#define REGISTER_12_BYTE_SIZE 4
#define REGISTER_12_OFFSET 0x50

#define REGISTER_13_BIT_FIELD_0_WIDTH 1
#define REGISTER_13_BIT_FIELD_0_MASK 0x1
#define REGISTER_13_BIT_FIELD_0_OFFSET 0

#define REGISTER_13_BYTE_WIDTH 4
#define REGISTER_13_BYTE_SIZE 4
#define REGISTER_13_OFFSET 0x54

#define REGISTER_13_BYTE_WIDTH 4
#define REGISTER_13_BYTE_SIZE 128
#define REGISTER_13_OFFSET 0x80

#endif
```

mcejp, 2022-06-11:
If it helps, you can look at [cheby](https://gitlab.cern.ch/cohtdrivers/cheby) for inspiration -- it is a tool with the same purpose as Rggen, and already implements generation of C headers.

taichi-ishitani, 2022-06-11:
Hi @mcejp ,
Thank you for your advice!