rggen/rggen #121: Issue with AXI write then read back.  Some reads return unknowns.

asicforever, 2022-09-15:
I have encountered an issue with RgGen generated code in our random regressions writing to 2 adjacent registers and subsequently reading back those values.  Some timings work correctly and some fail returning unknowns even though locations have been written to and read back.  

Here is a waveform screen shot:
[image]

Thanks for any insight.

Mike

taichi-ishitani, 2022-09-16:
Hi @asicforever ,
Thank you for your reporting this error.
I will investgate it.

For quick investigation, I'd like to reproduce this error on my side.
Can you reproduce this error by using this sample testbench and provide the test sequence?
https://github.com/rggen/rggen-sample-testbench

taichi-ishitani, 2022-09-16:
If it's difficult to reproduce the error, can you send a VCD file of your CSR block generted by RgGen?

asicforever, 2022-09-16:
Thanks for the quick response.  Yes we are using the generated SystemVerilog RTL.  I will do this when I get some cycles:
1. Try to enable the SVA checker.
2. Send you a VCD file.
3. Look at reproducing the error using https://github.com/rggen/rggen-sample-testbench.

Thanks

taichi-ishitani, 2022-09-16:
I've found a bus inside the AXI adapter module and fixed it. I think this bug also causes your issue.
rggen/rggen-sv-rtl@58d29f5db9b1912a249ec0b28ce1c6e7c988c22c

Can you get the latest common SV modules and try again?

asicforever, 2022-09-16:
I will do as you suggest first.

asicforever, 2022-09-16:
I just ran a previously failing simulation and now the simulation passes.  So this apparently fixed the issue.

Our verification group will hammer this code to make sure this issue is completely gone.  

Thanks for the quick help.
Mike

closed 2022-09-17 by taichi-ishitani