rggen/rggen #134: Citing RgGen

m-kru, 2023-02-16:
Is there any publication on RgGen that one can cite in papers, or anything else that you would like one to reference except GitHub rggen url?

taichi-ishitani, 2023-02-16:
Hi @m-kru ,

There are publications but these are written in Japanese and I'm not sure that these are suitable for your purpose.

* https://qiita.com/taichi-ishitani/items/5155b2928b7d85370ae6
* https://speakerdeck.com/taichi730/csr-wozi-dong-sheng-cheng-suru
* https://speakerdeck.com/taichi730/rggen-goshao-jie

taichi-ishitani, 2023-02-16:
* https://www.researchgate.net/publication/355988403_Control_and_Diagnostics_System_Generator_for_Complex_FPGA-Based_Measurement_Systems
* https://trepo.tuni.fi/bitstream/handle/10024/130330/Jyr%E4Kalle.pdf?sequence=2

These papers refer RgGen by using URL to GitHub/Librecores.

taichi-ishitani, 2023-02-20:
@m-kru ,

Can you close this issue if you have no further topics on this issue?

closed 2023-02-21 by m-kru