rggen/rggen #135: Register without bit-fields

m-kru, 2023-02-25:
How to define control register without bit-fields? All examples I have encountered contain bit-fields within the register. However, sometimes (most often?) there is only single bit-field with the same width as the register width.

The following works
```
  register {
    name 'reg0'
    bit_field { name 'foo'; bit_assignment width: 32; type :rw  }
  }
```
but is the following one also supposed to work?
```
  register {
    name 'foo'
    width 32
    type rw
  }
```
I also would like to omit the width if the field data occupies whole register.

taichi-ishitani, 2023-02-25:
RgGen does not support registers without bit fields.
Therefore, you need to define a bit field even if a register has single bit field only.

Implementation of RgGen supposes that a register must have bit fields so a lot of changes is needed to support such feature. So I don't intend to implement such feature.

taichi-ishitani, 2023-02-27:
@m-kru ,
Can you close this issue if you have no further topics?

closed 2023-02-27 by m-kru