rggen/rggen #149: Constants and expressions support

m-kru, 2023-04-26:
Does RgGen support constants and expressions? For example, I would like to define two module constants `X` and `Y`, and then I would like to calculate different values utilizing these constants in the expressions.

taichi-ishitani, 2023-04-26:
Can you tell me more details?
I'd like to know register map snippet which you want to write.

m-kru, 2023-04-27:
Pseudo description:
```yaml
register_blocks:
  - constants: {X: 1, Y: 2}
  - name: block_0
    byte_size: 256
    registers:
    - name: register_0
      bit_fields:
      - { name: bit_field_0, bit_assignment: { width: 4 }, type: rw  , initial_value: X ** Y }

```

taichi-ishitani, 2023-04-27:
You can do this by writing your register map by Ruby format.

```ruby
x = 1
y = 2
register_block {
  name 'block_0
  byte_size 256

  register {
    name 'register_0'
    bit_field { name 'bit_field_0'; bit_assignment { width: 4 }; type :rw; initial_value x**y }
  }
}
```

But other formats such as YAML, Excel, etc, do not support.

closed 2023-04-27 by m-kru