rggen/rggen #158: Vivado Doesn't like this systemverilog code

megeed, 2023-08-14:
`ifndef rggen_connect_bit_field_if
  `define rggen_connect_bit_field_if(RIF, FIF, LSB, WIDTH) \
  assign  FIF.valid                 = RIF.valid; \
  assign  FIF.read_mask             = RIF.read_mask[LSB+:WIDTH]; \
  assign  FIF.write_mask            = RIF.write_mask[LSB+:WIDTH]; \
  assign  FIF.write_data            = RIF.write_data[LSB+:WIDTH]; \
  assign  RIF.read_data[LSB+:WIDTH] = FIF.read_data; \
  assign  RIF.value[LSB+:WIDTH]     = FIF.value;
`endif
`ifndef rggen_tie_off_unused_signals
  `define rggen_tie_off_unused_signals(WIDTH, VALID_BITS, RIF) \
  if (1) begin : __g_tie_off \
    genvar  __i; \
    for (__i = 0;__i < WIDTH;++__i) begin : g \
      if ((((VALID_BITS) >> __i) % 2) == 0) begin : g \
        assign  RIF.read_data[__i]  = 1'b0; \
        assign  RIF.value[__i]      = 1'b0; \
      end \
    end \
  end
`endif

First one I got rid of the \ and made it one line.  It was happy.  The second one I couldn't get it to work.

taichi-ishitani, 2023-08-14:
Vivado can synthesis generated SV RTL so I don't understand what is wrong.
Can you put your synthesis log?

taichi-ishitani, 2023-08-15:
I'm using generated SV RTL with Vivado and no errors are reported.
So I think Vivado you are using is too old.

megeed, 2023-08-15:
Hello,

I think I got it down to a problem with size of the register(s).

If I remove this register.  It will load the file.  BTW I tried in vivado
22.2 and 23.1

generate if (1) begin : g_bd_iprrs_rbus
    genvar i;
    for (i = 0;i < 3;++i) begin : g
      rggen_bit_field_if #(6272) bit_field_if();
      `rggen_tie_off_unused_signals(6272,
6272'hffffffffffffffffffffffffffffffffffffffffffffffff000001ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff07ffffffffffffff00000fff0000007fffffffffffffffffffffffff000000ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff00001fffffffffffffffffff,
bit_field_if)
      rggen_default_register #(
        .READABLE       (1),
        .WRITABLE       (1),
        .ADDRESS_WIDTH  (12),
        .OFFSET_ADDRESS (12'(784*i)),
        .BUS_WIDTH      (32),
        .DATA_WIDTH     (6272),
        .VALUE_WIDTH    (6272)
      ) u_register (
        .i_clk        (i_clk),
        .i_rst_n      (i_rst_n),
        .register_if  (register_if[i]),
        .bit_field_if (bit_field_if)
      );

megeed, 2023-08-15:
I think we can close the issue.  This is definitely not a rggen issue.

taichi-ishitani, 2023-08-17:
I think you defined such big register to pack bit fields related to a functionality.
If yes, you can divide this register into multiple small registers and group them by using the `register_file` layer.
https://github.com/rggen/rggen-sample/blob/ebed8c251d31a3bed4e148168aa6b6e25283d74a/block_1.yml#L8