rggen/rggen #159: block_0.xlsx can't change register name

jamesningd, 2023-08-23:
**Describe the bug**
If I change the register name of register_0 and register_1 to reg_0 and reg_1, it will have error, but for other register, it is ok.

**To Reproduce**
Steps to reproduce the behavior:
1. just change the name of register_0 and register_1 in block_0.xlsx 
2. use following command in rggen-sample-master:  rggen -c config.yml --plugin rggen-verilog --plugin rggen-vhdl -o out block_0.xlsx

**Expected behavior**
I want to change the register name in excel like register_3, this can make my design much convenient.

taichi-ishitani, 2023-08-24:
Some bit fields refer bit fields of `register_0`. For example: `register4.bit_field_1`
[image]

To fix this error, you also need to update column `J` and change `regiter_0`/`register_1` to `reg_0`/`reg_1`.

closed 2023-08-25 by taichi-ishitani

jamesningd, 2023-08-25:
Thank you Taichi. It works now.