rggen/rggen #166: Support SPI as bus interface

hpretl, 2023-09-28:
**Is your feature request related to a problem? Please describe.**
It would be a super feature if `rggen` could be used to create an SPI-based register file.

**Describe the solution you'd like**
In addition to `apb` and `wishbone` also support `spi` type.

**Describe alternatives you've considered**
n/a

**Additional context**
Can provide this on request.

taichi-ishitani, 2023-09-28:
How about creating your own RgGen plugin to add SPI support?

hpretl, 2023-10-01:
I was thinking about it, but so far, I am shying the learning curve and the effort. Maybe we start a small project implementing it.

taichi-ishitani, 2023-10-02:
I created a plugin template.
https://github.com/taichi-ishitani/rggen-spi

By uisng this plugin, you only have to implement RTL adapter modules below.

* SV RTL: https://github.com/taichi-ishitani/rggen-spi/blob/master/rtl/rggen_spi_adapter.sv
* Verilog RTL: https://github.com/taichi-ishitani/rggen-spi/blob/master/rtl/rggen_spi_adapter.v
* VHDL RTL: https://github.com/taichi-ishitani/rggen-spi/blob/master/rtl/rggen_spi_adapter.vhd

You need to tell the path to the repository by using the `--plugin` switch:

```
$ rggen --plugin rggen-verilog --plugin rggen-vhdl --plugin PATH/TO/REPO/lib/rggen/spi.rb -c your/config.yaml your/register_map.yaml
```

If you don't need to generate Verilog RTL/VHDL RTL, you need to remove the realted plugin code below.

* https://github.com/taichi-ishitani/rggen-spi/blob/cf3eadd300c4b6101473b57da2c8415812634bb8/lib/rggen/spi/register_block/protocol/spi.rb#L23-L40
* https://github.com/taichi-ishitani/rggen-spi/blob/cf3eadd300c4b6101473b57da2c8415812634bb8/lib/rggen/spi/register_block/protocol/spi.rb#L42-L59

taichi-ishitani, 2023-10-02:
See https://github.com/rggen/rggen/wiki/Specification-for-RgGen-Bus-Interface-Protocol for the protocol specification of the `rggen_bus_if`.

taichi-ishitani, 2023-10-10:
@hpretl 
Does the plugin template work well for you?

hpretl, 2023-10-10:
We have not yet looked into it, but it is on our todo list.

taichi-ishitani, 2023-10-10:
Ok, I understood.