rggen/rggen #169: Error while starting the uvm_reg_mem_hdl_paths_seq.

Tejoyadav, 2023-11-27:
Hi Taichi,

I'm able to run my RAL testcase by starting uvm_reg_hw_reset_seq and uvm_reg_access_seq. But when I'm trying to start uvm_reg_mem_hdl_paths_seq, the following errors are comming for every register.

[image]

I'm thinking like if the hdl path is wrong, then above two sequences should have created the issues. Suggest me, should I ignore these errors?

Regards, 
Tejoyadav

taichi-ishitani, 2023-11-27:
Generated RAL models do not support PLI based backdoor access so I think you can ignore these errors.

Tejoyadav, 2023-11-27:
Okay, thankyou for confirming

closed 2023-11-27 by Tejoyadav

taichi-ishitani, 2023-11-28:
PLI is a specification of software interface.
By using PLI, you can access/update Verilog objects (e.g. `reg` variables) from oher programing langurage such as C.
By default, UVM bacdoor access feature is implemented by using PLI to allow hierchical access using string variable.

PLI is a useful feature but it may cause slower simulation runtime.
To avoid this problem, I introduce virtual interface based backdoor access feature.