rggen/rggen #190: Implement: Parameter-Driven Bit width assignment in SV/Verilog files from input file formats

pratheekjain, 2024-01-25:
**Implementation of parameter usage for Bit width assignment in input file formats and corresponding generation in System Verilog and Verilog Files**

In the current implementation, the bit width assignment is being used in the input format (XLSX) as hard coded values. This has been identified as a potential issue as it would be more appropriate to use the 'PARAMETER' instead. The expected behavior is to use the 'PARAMETER' in place of hardcode values for bit assignment in the input format and generate the 'PARAMETER' in the System Verilog (SV) and Verilog files.
This change will ensure that the 'PARAMETER' is consistently used across all files, improving the readability, reusability and maintainability of the code.
 
I have spent quite a time trying to work this out with no success, so I thought I would ask for your help. This requires a feature enhancement in the Ruby script to support the ‘PARAMETER’, ensuring its correct usage in SV and Verilog files.

taichi-ishitani, 2024-01-25:
Hi @pratheekjain ,

Thank you for opening the issue!
It's difficult to add such feature to an input file formats because it will need a lot of changes.

You can do use variables on your register map files if you can select Ruby file as an input file like below.

```ruby
width = 4
register_block {
  name 'block_0'
  byte_size 256

  register {
    name 'register_0'
    bit_field { name 'bit_field_0'; bit_assignment width: width; type :rw; initial_value 0 }
    bit_field { name 'bit_field_1'; bit_assignment width: width; type :rw; initial_value 0 }
  }
}
```

But generated RTL does not have parameters showing such variables.

taichi-ishitani, 2024-01-31:
@pratheekjain ,
Can you close this issue?

pratheekjain, 2024-01-31:
> Hi @pratheekjain ,
> 
> Thank you for opening the issue! It's difficult to add such feature to any input file formats because it will need a lot of changes.
> 
> You can do use variables on your register map files if you can select Ruby file as an input file like below.
> 
> ```ruby
> width = 4
> register_block {
>   name 'block_0'
>   byte_size 256
> 
>   register {
>     name 'register_0'
>     bit_field { name 'bit_field_0'; bit_assignment width: width; type :rw; initial_value 0 }
>     bit_field { name 'bit_field_1'; bit_assignment width: width; type :rw; initial_value 0 }
>   }
> }
> ```
> 
> But generated RTL does not have parameters showing such variables.

Okay, Thank you @taichi-ishitani .

pratheekjain, 2024-01-31:
> @pratheekjain , Can you close this issue?

Okay

closed 2024-01-31 by pratheekjain