rggen/rggen #196: [Request] Add support for library name

SzymonHitachi, 2024-02-22:
When building rggen standalone through. e.g. [tsfpga](https://github.com/tsfpga/tsfpga) build system, by default rggen is put in a library named `rggen`. This is not an issue when referencing internal modules, however the generated final csr block will need to use the library name it was compiled to.

To achieve this, the output file of register block needs to have changed `use work.rggen_rtl.all;` in a way that instead of `work`, another library name can be passed. Either through command line or input file.

Using a library also requires a library statement.
E.g.

if using `work` by default keep

`use work.rggen_rtl.all`

however if using a library e.g. `SOME_LIB`

then the header requires to be:

```vhdl
library SOME_LIB;
use SOME_LIB.rggen_rtl.all;
```

Additionally the entity calls will need to be updated to:

`entity SOME_LIB.rggen_axi4lite_adapter`

This change is relevant only for the top block modules instantiated in user design.

taichi-ishitani, 2024-02-22:
Common modules also use the library `work` like below.
https://github.com/rggen/rggen-vhdl-rtl/blob/219fdd5a1d38f043b8e42fca3ee14e9ba905f1cf/rggen_axi4lite_adapter.vhd#L263

Do we need to change these line?

SzymonHitachi, 2024-02-22:
In theory you do not need to because `work` is not exactly a library. It's a designator of `this` or `self` which means "search the current scope I am within". so if we compile every rggen file with `work.` in `SOME_LIB` library, then they will reference whatever is in `SOME_LIB` only. It's the outermost (so the ones that are included in user project and call `SOME_LIB.rggen`) modules that require this.

taichi-ishitani, 2024-02-23:
Hi @SzymonHitachi ,
I've pushed commits for this feature to the `add_library_name_property` branch.
Can you check these commits work well on your side?

```
$ git clone https://github.com/rggen/rggen-vhdl.git
$ cd rggen-vhdl
$ git pull add_library_name_property
$ rggen --plugin ./lib/rggen/vhdl.rb --enable vhdl --o out -c path/to/your/conig.yaml path/to/your/regster_map.yaml
```

taichi-ishitani, 2024-02-27:
You don't need to change your XSLX register map but need to update your config file like below.

```yaml
address_width: 16
bus_width: 32
protocol: apb
library_name: SOME_LIB # add this line
```

SzymonHitachi, 2024-02-27:
So the library change works, but the code is no longer synthesizing:
[image]
With the error:
```[Synth 8-10097] type error near i ; current type integer; expected type unresolved_unsigned``` 
Which seems to be related to the below changes:
[image]

taichi-ishitani, 2024-02-27:
Can you change this line like below and try again?

```vhdl
OFFSET_ADDRESS => x"0"+4*i
```

SzymonHitachi, 2024-02-27:
Fixes the previous issue but now somethings off with address decoder?
[image]

for CSR with `  ADDRESS_WIDTH: positive := 8;`

taichi-ishitani, 2024-02-27:
How about this?

```vhdl
OFFSET_ADDRESS => x"00"+4*i
```

SzymonHitachi, 2024-02-27:
This one seems to fix the issues, probably due to vector size.

taichi-ishitani, 2024-02-27:
Thanks for confirming.
I will open a ticket to fix this error.

closed 2024-02-27 by taichi-ishitani

taichi-ishitani, 2024-02-28:
@SzymonHitachi ,

I've released rggen-vhdl plugin v0.10.0. This version includes fixes for #196 and #197.
Can you update the plugin and confirm if your issue is resolved?

How to update:

```
gem update -NE rggen-vhdl
```