rggen/rggen #198: rc, w0c, w1c, wc, woc: bit field set when it should be cleared?

ifkato, 2024-03-05:
**Describe the bug**

In the bit field implementation of rc, w0c, w1c, wc, and woc ("read/write something to clear bit"), it seems like "set" signal is connected rather than "clear".

**To Reproduce**

Steps to reproduce the behavior:
1. specify "rc", "w0c", "w1c", "wc" or "woc" as bit field type.
2. RTL file generated with "set" signal connected.

**Expected behavior**

RTL file generated with "clear" signal connected.

**Screenshots**

n/a

**Additional context**

related code:

- rggen-systemverilog: https://github.com/rggen/rggen-systemverilog/blob/c731542f571aefcd8d064aa91ad7445619286a87/lib/rggen/systemverilog/rtl/bit_field/type/rc_w0c_w1c_wc_woc.erb
- rggen-verilog: https://github.com/rggen/rggen-verilog/blob/2833672a1a06d46fda745d4711bec5815b7f001c/lib/rggen/verilog/rtl/bit_field/type/rc_w0c_w1c_wc_woc.erb#L20

Also, [the document](https://github.com/rggen/rggen/wiki/Register-Map-Specifications#type-1)  section about w1c may need to be fixed.

大変便利なソフトウェアを作ってくださり、ありがとうございます！

ifkato, 2024-03-06:
なるほどです！私の誤解だったようですね。
ありがとうございます！

closed 2024-03-06 by ifkato

taichi-ishitani, 2024-03-06:
質問や、機能追加の要望などありましたら、お気軽に issue を建ててくださいませ。