rggen/rggen #207: [external] register type seems to not be working

SzymonHitachi, 2024-05-10:
It seems that external registers mode does not work 
```vhdl
Fatal: (vsim-3420) Array lengths do not match. Left is 11 (10 downto 0). Right is 12 (11 downto 0).
#    Time: 473200 ps  Iteration: 1  Process: csr_i/g_DATA/u_register/line__119 File: rggen/rggen_external_register.vhd
# Fatal error in Process line__119 at rggen/rggen_external_register.vhd line 47
```
failing line:
```vhdl
address := std_logic_vector(unsigned(register_address) - START_ADDRESS);
```
external register is defined with offset `0x10` (determined byregisters existing before it)
[image]

version:
```
$ rggen --plugin rggen-vhdl --verbose-version
RgGen 0.33
  - rggen-core 0.33.0
  - rggen-default-register-map 0.33.0
  - rggen-sv-rtl 0.33.0
  - rggen-sv-ral 0.33.0
  - rggen-c-header 0.5.0
  - rggen-markdown 0.26.0
  - rggen-spreadsheet-loader 0.25.2
  - rggen-vhdl 0.10.0
  ```

SzymonHitachi, 2024-05-10:
Issue occurs because instantiation of `rggen_external_register` in the autogenerated code
```vhdl
    u_register: entity work.rggen_external_register
      generic map (
        ADDRESS_WIDTH => 11,
        BUS_WIDTH     => 32,
        STROBE_WIDTH  => DATA_STROBE_WIDTH,
        START_ADDRESS => x"010",
        BYTE_SIZE     => 1024
      )
```

doesn't provide the size of string, thus it's by default expanded to 12 in `START_ADDRESS`      

```vhdl
    u_register: entity work.rggen_external_register
      generic map (
        ADDRESS_WIDTH => 11,
        BUS_WIDTH     => 32,
        STROBE_WIDTH  => DATA_STROBE_WIDTH,
        START_ADDRESS => 11x"010",
        BYTE_SIZE     => 1024
      )
```
Specifying the actual size represented by string START_ADDRESS => **11**x"010", fixes the issue.

taichi-ishitani, 2024-05-10:
Hi @SzymonHitachi ,
Thank you for your reporting this issue.

I've pushed the fix for this error to the branch below.
https://github.com/rggen/rggen-vhdl-rtl/tree/fix_width_mismatch
Can you checkout this branch and confirm this error is resolved?

taichi-ishitani, 2024-05-15:
Hi @SzymonHitachi ,
Does this change resolve your issue?

closed 2024-05-22 by taichi-ishitani

SzymonHitachi, 2024-05-22:
> Hi @SzymonHitachi , Does this change resolve your issue?

I apologize for late response. It did indeed slip my mind. Yes this one fixes the issue.