rggen/rggen #216: rggen-vhdl generates APB interface when the protocol is specified as wishbone

m-kru, 2024-07-12:
`rggen-vhdl` generates APB interface when the protocol is specified as wishbone.

```yaml
bus_width: 32
address_width: 32
protocol: wishbone
```

```vhdl
entity Main is
  generic (
    ADDRESS_WIDTH: positive := 8;
    PRE_DECODE: boolean := false;
    BASE_ADDRESS: unsigned := x"0";
    ERROR_STATUS: boolean := false;
    INSERT_SLICER: boolean := false
  );
  port (
    i_clk: in std_logic;
    i_rst_n: in std_logic;
    i_psel: in std_logic;
    i_penable: in std_logic;
    i_paddr: in std_logic_vector(ADDRESS_WIDTH-1 downto 0);
    i_pprot: in std_logic_vector(2 downto 0);
    i_pwrite: in std_logic;
    i_pstrb: in std_logic_vector(3 downto 0);
    i_pwdata: in std_logic_vector(31 downto 0);
    o_pready: out std_logic;
    o_prdata: out std_logic_vector(31 downto 0);
    o_pslverr: out std_logic;
```

taichi-ishitani, 2024-07-12:
I was not able to reproduce this issue on my side.
I think you forgot to specify your config file.
Can you confirm that?

m-kru, 2024-07-12:
What does you `config.yaml` look like?

taichi-ishitani, 2024-07-12:
Same with the config file which you show.

```yaml
bus_width: 32
address_width: 32
protocol: wishbone
```

m-kru, 2024-07-12:
This snippet is a part of my register description file. I do not provide any config file.

taichi-ishitani, 2024-07-12:
You need to put these properties to your config file.

m-kru, 2024-07-12:
Why? Why do I need to duplicate this information?

taichi-ishitani, 2024-07-12:
You don't have to put them to your register map file.

m-kru, 2024-07-12:
If I put them there `rggen` doesn't issue any error or warning. However, they don't work. I feel confused.

m-kru, 2024-07-13:
https://github.com/rggen/rggen/issues/218

m-kru, 2024-07-13:
What version of rggen do you use? I use RgGen 0.33 and even if I put this data into the config file, I still get output files utilizing APB, not Wishbone.

taichi-ishitani, 2024-07-14:
I use 0.33.
Can you provide me your config file and execution command?

m-kru, 2024-07-14:
The source of the problem was lack of the `-c` argument.

```
rggen config.yaml --plugin rggen-vhdl bus.yaml 
```

Looks like rggen accepts invalid configuration in multiple places and simply silently moves on.

taichi-ishitani, 2024-07-14:
> Looks like rggen accepts invalid configuration in multiple places and simply silently moves on.

This will be fixed by #218 .

taichi-ishitani, 2024-07-20:
I released RgGen v0.33.3.
From this release, RgGen will report an error when unknown field is given like below.
[image]

closed 2024-07-20 by taichi-ishitani