rggen/rggen #219: Add variable placeholders for comments

SzymonHitachi, 2024-07-15:
When generating multiple copies of same register, I would like to be able to have the iteration number of it as a variable that could be used in comment, e.g.
[image]
In the comment I would be able to have `Interrupt {i}` (syntax is just an example) and each of the generated registers will have a comment for:
```
Interrupt 0
Interrupt 1
Interrupt 2
Interrupt 3
```

Instead of 
```
Interrupt
```
four times

taichi-ishitani, 2024-07-16:
How about processing a comment string as an ERB template?
https://en.wikipedia.org/wiki/ERuby

For exmaple:

```erb
<% register.size.times do %>
Interrupt <%= i %>
<% end %>
```

SzymonHitachi, 2024-07-16:
This syntax could work (currently it seems to be ignored when added to the input comment section), yes but for the mentioned config, only a single register in the MD is generated with 4 offsets:
[image]
An option to produce 4 registers would probably also be needed for this to work as I described?

taichi-ishitani, 2024-07-16:
You want to expand an array register into each array elements when generating Markdonw?

SzymonHitachi, 2024-07-16:
Hmmm I think that would be it i guess? This way each of the elements will get it custom description in the MD

taichi-ishitani, 2024-07-16:
OK.
I will consider how to implement it.

taichi-ishitani, 2024-07-17:
> You want to expand an array register into each array elements when generating Markdonw?

It will take a lot of changes to implement this feature.

So, for now, I will implement template support on comment fields.
Is this template feature suitable for what you want to do?

closed 2025-01-21 by taichi-ishitani