rggen/rggen #250: rggen sample plugin is broken with v0.35.0

alejandro-castro-ortegon, 2025-03-04:
**Describe the bug**
Trying to run the [rggen sample plugin](https://github.com/rggen/rggen-sample-plugin) is no longer possible.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the sample repo located [here](https://github.com/rggen/rggen-sample-plugin)
2. `cd` into the downloaded repo
3. run `rggen --plugin ./lib/rggen/my_plugin.rb --enable sv_rtl --output out sample/sample.yaml`
4. See error

**Expected behavior**
Obtain the output files, the same files located [here](https://github.com/rggen/rggen-sample-plugin/tree/master/out).

**Error message**
```
[NameError] undefined local variable or method `array_port_format' for #<#<Class:0x00007fafbbb54600>:0x00007fafbb6d2168>
```

**Additional context**
Rggen version where the issue is present: v0.35.0

taichi-ishitani, 2025-03-04:
Hi @alejandro-castro-ortegon ,

Thank you for reporting this issue!
I've fixed the sample plugin. Could you upate the plugin and try again?

taichi-ishitani, 2025-03-04:
Feel free to open a new issue if you have any questions!