rggen/rggen #262: Bit-for-bit Masking feature

taichi-ishitani, 2025-09-02:
Add a new bit field type same with the following.
https://github.com/hdl-registers/hdl-registers/issues/244#issue-3111818654

taichi-ishitani, 2025-09-02:
@chili-chips-ba
Thanks for an idea for a new feature!

chili-chips-ba, 2025-09-02:
Here is another one:
- https://github.com/chili-chips-ba/wireguard-fpga/issues/10

chili-chips-ba, 2025-09-02:
Yes, that would be very useful!

If it is not too much work, please try to replicate the tests that come with UVM, including things such as:
  - address aliasing
  - data aliasing

chili-chips-ba, 2025-09-04:
That's the standard feature of all professional, closed-source CSR-Gen flows we've worked with. It is meant to ring out both **integration** and **manufacturing / build** errors.

We've seen it pay off in more than one case. It would also set the RgGen flow apart from all other open-source solutions.

taichi-ishitani, 2025-09-16:
* Implement this as a register type but not a bit field type
* Treat lower half bytes as a write data and upper half bytes as a bit mask
    * bit fields in this register should be located in its lower bytes